sairon/esphome-nspanel-lovelace-ui #2: Does not boot

vstonkus, 2022-05-22:
ESPhome does not boot after flashing neither it outputs any logs via connected serial.

The config I am using:
`esphome:
  name: nspanel

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "**"

ota:
  password: "**"

wifi:
  ssid: "**"
  password: "**"

external_components:
  - source:
      type: git
      url: https://github.com/sairon/esphome-nspanel-lovelace-ui
      ref: dev
    components: [ nspanel_lovelace ]

uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200
  id: tf_uart

mqtt:
  id: mqtt_client
  broker: 10.0.10.125
  on_message:
    topic: cmnd/nspanel/CustomSend
    then:
      - lambda: |-
          id(nspanel).send_custom_command(x);

nspanel_lovelace:
  id: nspanel

  on_incoming_msg:
    then:
    - mqtt.publish_json:
        topic: tele/nspanel/RESULT
        payload: |-
          root["CustomRecv"] = x;
    `

sairon, 2022-05-24:
Do you get at least the ROM message from the ESP32 on device reset? It looks like this:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
...
```

If you don't get  even this message, then the problem is likely HW related - either something wrong is with the wiring or with device in general. What method have you used to flash the firmware initially?

visvaldasstonkus, 2022-05-24:
I forgot to detach GPIO0 from the ground when I was checking serial output. So it does output, but seems to be in a boot loop. Rebooting after "Setting up WiFi..." @sairon do you have any idea?

```
[13:29:53]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:29:53]configsip: 0, SPIWP:0xee
[13:29:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:29:53]mode:DIO, clock div:2
[13:29:53]load:0x3fff0018,len:4
[13:29:53]load:0x3fff001c,len:1044
[13:29:53]load:0x40078000,len:8896
[13:29:53]load:0x40080400,len:5828
[13:29:53]entry 0x400806ac
[13:29:54][I][logger:242]: Log initialized
[13:29:54][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[13:29:54][D][esp32.preferences:114]: Saving preferences to flash...
[13:29:54][D][esp32.preferences:131]: NVS data not changed skipping 233825511  len=4
[13:29:54][I][app:029]: Running through setup()...
[13:29:54][C][uart.arduino_esp32:077]: Setting up UART...
[13:29:54][C][wifi:037]: Setting up WiFi...
[13:29:54]ets Jul 29 2019 12:21:46
[13:29:54]
[13:29:54]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:29:54]configsip: 0, SPIWP:0xee
[13:29:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:29:54]mode:DIO, clock div:2
[13:29:54]load:0x3fff0018,len:4
[13:29:54]load:0x3fff001c,len:1044
[13:29:54]load:0x40078000,len:8896
[13:29:54]load:0x40080400,len:5828
[13:29:54]entry 0x400806ac
[13:29:54][I][logger:242]: Log initialized
[13:29:54][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[13:29:54][D][esp32.preferences:114]: Saving preferences to flash...
[13:29:54][D][esp32.preferences:131]: NVS data not changed skipping 233825511  len=4
[13:29:54][I][app:029]: Running through setup()...
[13:29:54][C][uart.arduino_esp32:077]: Setting up UART...
[13:29:54][C][wifi:037]: Setting up WiFi...
[13:29:55]ets Jul 29 2019 12:21:46
[13:29:55]
[13:29:55]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

sairon, 2022-05-24:
Judging from line:
```[13:29:54]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)```
I'd say it's a power supply issue - the device doesn't get enough power when it starts to search for available WiFi networks and the voltage drops too low even before it could detect a brownout event.

Either try using a different power supply or simply connect the original relay/PSU board and check if the device connects to WiFi. If you don't have a beefy 3V3 supply, you can use a second 5V PSU connected to the interconnection pins - see [pinout on this page](https://blakadder.com/nspanel-teardown/), just remember to connect GND from the second PSU and the USB-serial.

Just don't try to use the original PSU board with a serial attached - IIRC it's not isolated and you could fry it or the attached PC as well.

vstonkus, 2022-05-24:
After connecting it via the exposed pins using 5V instead of 3V3 on the serial connectors I got it working. Thank you.

Also I had to add following config I found [here](https://github.com/esphome/esphome/pull/2702) to power up the screen.
```
switch:
  - platform: gpio
    name: Screen Power
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON
```

I got it fully functioning. Thanks again for the help to figure it out. It feels a bit faster than Tasmota and allows me to add other components into esphome like connecting it to my whole home canbus for high availability communication between the nspanel and esphome based led dimmers (gonna use nspanel hardware buttons for high availability control). Would wish if entire nspanel lovelace ui could run locally inside the esp instead of rely on appdaemon being online. Appdaemon could be responsible for receiving states only and maybe push updates to nspanels when changes happen in HA layout.

Btw. native api over MQTT would be nice too, since we have esphome.

closed 2022-05-24 by vstonkus