sairon/esphome-nspanel-lovelace-ui #5: Working Correctly

OmriShiv, 2022-07-12:
Hello,
I really appreciate taking the time to make this compatible with ESPHome. I think I have everything set up correctly? I can see MQTT messages, I have appdaemon running and the display is able to get the time and date; however, sending commands other than a page for my lights doesn't seem to work. Using the app.yaml, I don't think my changes are being persisted (no cards, screen timeout doesn't change, can't seem to get the weather on it).

I may have screwed up somewhere, I'm using HA and had to separately install appdaemon. Wondering if you might have some guidance as to why the cards I've set up in the yaml are not loading. 

```
nspanel:
  module: nspanel-lovelace-ui
  class: NsPanelLovelaceUIManager
  config:
    panelRecvTopic: "tele/nspanel/RESULT"
    panelSendTopic: "cmnd/nspanel/CustomSend"
  sleepTimeout: 20
  screensaver:
    entity: weather.home
  cards:
    - type: cardEntities
      title: Lights
      entities:
        - light.office
    - type: cardAlarm
      title: Alarm
      entity: alarm_control_panel.hsgw
    - type: cardMedia
      entity: media_player.move
```

I can use the HA service to sent `pageType~cardThermo~themostat~climate.ecobee_2` and get the card displayed, but it disappears very quickly and also doesn't have an entity. Same with the alarm and media cards

Aldaran, 2022-07-13:
> I may have screwed up somewhere, I'm using HA and had to separately install appdaemon. Wondering if you might have some guidance as to why the cards I've set up in the yaml are not loading.

It works for me... fine. I see a slowdown, but this is the specifics of the AppDaemon backend and TFT driver due to heavy workload. But you can use.

After touching, your cards with entities (Lights / Alarm / Media) should appear on your screen.
If not, then there is some kind of error, I only had this on Tasmota, when I ran the wrong script responsible for loading the driver via 'autoexec.be' and when I uploaded the developer firmware to the nextion.
"pageType~cardThermo~" causes just an empty thermostat card, then you need to call your entity update...:
entityUpd,*internalNameEntiy*,*heading*,*currentTemp*,*destTemp*,*status*,*minTemp*,*maxTemp*,*stepTemp*
And this call will be on time, in fact it's like notifity...

I figured it out a little myself, it's very hard to do it without normal manuals...

For example:
  **- platform: gpio
    name: $friendly_name Relay 1
    id: relay_1
    pin:
      number: 22
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      - lambda: 'id($device_name).send_custom_command("pageType~popupNotify");'
      - lambda: |-
          if(id(relay_2).state) {
            id($device_name).send_custom_command("entityUpdateDetail~id~Light in Bedroom~65535~ON~3840~ON~3840~ ~65535~5");
          } else {
            id($device_name).send_custom_command("entityUpdateDetail~id~Light in Bedroom~65535~ON~3840~OFF~61440~ ~65535~5");
          }
    on_turn_off:
      # send to screensaver
      - lambda: 'id($device_name).send_custom_command("pageType~popupNotify");'
      - lambda: |-
          if(id(relay_2).state) {
            id($device_name).send_custom_command("entityUpdateDetail~id~Light in Bedroom~65535~OFF~61440~ON~3840~ ~65535~5");
          } else {
            id($device_name).send_custom_command("entityUpdateDetail~id~Light in Bedroom~65535~OFF~61440~OFF~61440~ ~65535~5");
          }**

In this case, the light can be called with one command:
**...
     on_turn_on:
       - lambda: 'id($device_name).send_custom_command("pageType~popupLight~Relay 1~light.ns_panel_1_relay_1~On");'**
 
PS. And I want to ensure that after five minutes my page is returned to page #1, and not in the state that was changed by my actions

OmriShiv, 2022-07-13:
Yes, I have that all the same other than setting the appdaemon.yaml and apps.yaml file in my appdaemon docker location, not the one in HA (since I run them separately). I wonder if I should move it and mount it from there... That's one test

OmriShiv, 2022-07-13:
Although, I'm hoping I didn't just break something... I tried to reupload the tft, but it's stalled

OmriShiv, 2022-07-13:
```
[09:18:27][D][nspanel_lovelace:098]: Sending: DRAKJHSUYDGBNCJHGJKSHBDN
[09:18:27][D][nspanel_lovelace:098]: Sending: recmod=0
[09:18:27][D][nspanel_lovelace:098]: Sending: recmod=0
[09:18:27][D][nspanel_lovelace:098]: Sending: connect
[09:18:27][D][nspanel_lovelace_upload:150]: Connected
[09:18:27][D][nspanel_lovelace_upload:156]: Requesting URL: http://nspanel.pky.eu/lui-us-p-release.tft
[09:18:27][D][nspanel_lovelace_upload:188]: Updating Nextion
[09:18:27][D][nspanel_lovelace:098]: Sending: whmi-wris 7667820,115200,1
[09:18:27][D][nspanel_lovelace_upload:206]: Waiting for upgrade response
[09:18:28][D][nspanel_lovelace_upload:211]: Upgrade response is [05]
[09:18:28][D][nspanel_lovelace_upload:214]: preparation for tft update done
[09:18:28][D][nspanel_lovelace_upload:238]: Allocating buffer size 65536, Heap size is 219740
[09:18:28][D][nspanel_lovelace_upload:255]: Updating tft from "http://nspanel.pky.eu/lui-us-p-release.tft" with a file size of 7667820 using 65536 chunksize, Heap Size 159712
[09:18:28][D][nspanel_lovelace_upload:032]: Requesting range: bytes=0-16383
[09:18:28][D][nspanel_lovelace_upload:081]: fetched 16384 bytes
[09:18:29][D][nspanel_lovelace_upload:090]: Uploaded 0.1 %, remaining 7663724 B
INFO 10.45.48.191: Ping timed out!
INFO Disconnected from ESPHome API for 10.1.1.189
WARNING Disconnected from API
```

Aldaran, 2022-07-13:
I don't even know about the location of the files in the docker, but if you opened the screensaver with the weather, then in theory this is the correct location ... Did it happen with the nextion update? 

Try to reinstall nspanel-lovelace-... choosing not the firmware version, but main

OmriShiv, 2022-07-13:
I tried the main one too that I'm hosting on my homeassistant; no luck. I'll keep trying

I have never managed to get the weather to display

Aldaran, 2022-07-13:
Does esphhome connect to mqtt?  Topics should appear nspanel/... with sensors/switch/other..

OmriShiv, 2022-07-13:
This is my apps.yaml:
```
 {
    "module": "nspanel-lovelace-ui",
    "class": "NsPanelLovelaceUIManager",
    "config": {
        "panelRecvTopic": "tele/nspanel/RESULT",
        "panelSendTopic": "cmnd/nspanel/CustomSend"
    },
    "screensaver": {
        "entity": "weather.home",
        "weatherUnit": "fahrenheit"
    },
    "model": "us-p"
} 
```

Aldaran, 2022-07-13:
[image]

I have it like this, I apologize for the picture, markdown pisses me off, I need to edit the code every time ...
Don't pay attention to the threads...

OmriShiv, 2022-07-13:
You can use a triple backtick to paste code. Other than the mqtt topics being reversed for the device (I assume you made the change as well in your esphome config), I'm not seeing a difference. I know the services work since I can send those from HA to the esp and it's displayed, so something with appdaemon must not be working properly. I can see it streaming the mqtt messages, but something is getting lost; I'll keep at it.

OmriShiv, 2022-07-13:
ARGH! I figured it out: of course it was yaml. The structure was incorrect, it wasn't indented properly... wow

Aldaran, 2022-07-13:
[nspanel1.txt](https://github.com/sairon/esphome-nspanel-lovelace-ui/files/9104525/nspanel1.txt)
My firmware esphome...

Aldaran, 2022-07-13:
There are not so many troubles with Tasmota, but I don’t like how much my NS Panel heats up, and it incorrectly shows the temperature of the device (although it may be correct, but esp32 warms up the sensor).

But here you are your own boss, you can add something, you can remove it ... for example, some services are either clearly not needed or do not work very correctly

OmriShiv, 2022-07-13:
Yes, I was hoping to stay with ESPHome, but I do feel the display is very sluggish even with just one card and a few entities. I also can't seem to get it into portrait mode

Aldaran, 2022-07-13:
> Yes, I was hoping to stay with ESPHome, but I do feel the display is very sluggish even with just one card and a few entities. I also can't seem to get it into portrait mode

There is such a project https://github.com/marcfager/nspanel-mf, but it provides a base, everything works smartly there, but any actions force you to rewrite the firmware code and work with pictures for the screen ... and there are not always enough skills for your wishes...

sairon, 2022-07-13:
Hi @OmriShiv, if I understand it correctly, it seems that your problem was in the misconfiguration of the AppDaemon service. With that being sorted out, I'm closing the issue. In general, this ESPHome component can't do much about UI snappiness - that entirely depends on the original **NSPanel Lovelace UI** architecture. If you want to discuss that, please head over to that project, or even better, to the [Unofficial Nextion Discord](https://discord.me/unofficialnextion).

For anyone who encounters similar issue in the future, the debugging steps should be first checking that your MQTT broker sees the messages sent by the display when you touch the panel, and then that it receives messages sent from the AppDaemon - for example time update every minute. If both are published to MQTT, it's likely caused by the configuration of the AppDaemon application or wrong topic names.

closed 2022-07-13 by sairon

OmriShiv, 2022-07-13:
Yes, thank you, it was entirely in my apps.yaml config wasn't indented properly I think. Thank you