sairon/esphome-nspanel-lovelace-ui #8: Status misses when entities change in Home Assistant

HA-TB303, 2022-08-13:
Hi,

I have been trying https://github.com/joBr99/nspanel-lovelace-ui with both Tasmota and your esphome component and I am having some issues with status change misses on the nspanel display.
It does not matter of the status changes from home assistant or from the nspanel itself.

Reproduction: Configure a grid panel with external lights from Home Assistant like:

```
nspanelbadk:
  module: nspanel-lovelace-ui
  class: NsPanelLovelaceUIManager
  config:
    panelRecvTopic: "tele/nspanelbadk/RESULT"
    panelSendTopic: "cmnd/nspanelbadk/CustomSend"
    locale: "en_EN"
    model: "eu"
    sleepTimeout: 120
    sleepBrightness:
      - time: "6:00:00"
        value: 8
      - time: "23:00:00"
        value: 2
    screensaver:
      entity: weather.dark_sky
    cards:
      - type: cardGrid
        title: Bathroom
        key: cardgrid1
        entities:
          - entity: light.kantoor_led_wand_boven
            name: Shower light
          - entity: light.kantoor_led_wand_onder
            name: Bath light
          - entity: light.kantoor_rechts
            name: Vanity light
          - entity: switch.awtrix
            name: Floor heating
          - entity: sensor.temperatuur_en_vochtsensor_badkamer_temperature
            name: Temperature
          - entity: sensor.temperatuur_en_vochtsensor_badkamer_humidity
            name: Humidity
      - type: cardMedia
        title: Bathroom
        entity: media_player.badkamer
      - type: cardThermo
        title: Bathroom
        entity: climate.panasonic_airco_tim_s_kamer
      - type: cardEntities
        title: Test Entities Card
        entities:
          - entity: light.kantoor_led_wand_boven
```
Then turn the light on and off from Home Assistant repeatedly and notice the misses. I have been debugging this with help from joBr99 here, bit no luck sofar: https://github.com/joBr99/nspanel-lovelace-ui/issues/392

This issues does not occur when using tasmota and the exact same config in adddeamon.

My ESPhome config (which is nearly identical to your example config)

```
substitutions:
  devicename: nspanelbadk
  friendly_devicename: NSPanel Badkamer
  panel_recv_topic: "tele/nspanelbadk/RESULT"
  panel_send_topic: "cmnd/nspanelbadk/CustomSend"

esphome:
  name: $devicename

esp32:
  board: esp32dev
[... 220 lines cut ...]
  - platform: gpio
    name: $friendly_devicename Left Button
    pin:
      number: GPIO14
      inverted: true
    id: left_button
    on_press:
      then:
        - switch.turn_on: relay_1

  # Right button below the display
  - platform: gpio
    name: $friendly_devicename Right Button
    pin:
      number: GPIO27
      inverted: true
    id: right_button
    on_press:
      then:
        - switch.toggle: relay_2
```

Short video that "proves" the issue:

https://user-images.githubusercontent.com/9609805/184479529-e59e42af-810f-4b1c-909c-5a6f439bea84.mp4

tc245, 2022-08-31:
I have the same issue though I have noticed that it only seems to affect light domain entities. Entities in the switch domain seem to work fine. I got around this by creating a [template switch in home assistant](https://www.home-assistant.io/integrations/switch.template/). This still switches the light entity on and off but adding to apps.yaml instead of the original light entity fixes the updating problem on the display for me.

HA-TB303, 2022-08-31:
Yes, you are right. I just tested again to verify and indeed only light entities are effected. Switch works fine.
Your solution is a nice work-around, but you can not dimm a switch:(

tc245, 2022-08-31:
> Yes, you are right. I just tested again to verify and indeed only light entities are effected. Switch works fine. Your solution is a nice work-around, but you can not dimm a switch:(

Yep, true. Hopefully the bug can be fixed, Odd that it only affects lights and not switches though.

HA-TB303, 2022-08-31:
Indeed. Hopefully [[sairon](url)](https://github.com/sairon) is still willing and able to check this bug.

sairon, 2022-08-31:
Thanks for your findings, folks, I probably have some anecdotal evidence of the same issue as well, yet I haven't done any debugging yet.  It's quite strange it only affects lights, so I'll look deeper into it, thanks for the hint. Can't promise any ETA, but it's on my radar.

tc245, 2022-09-01:
Thanks @sairon. Did some more experimenting last night... I think actually that the problem still remains with switch entities after all. Operating the switch from the panel seems to update as it should but if the switch is operated from elsewhere it doesn't update the state on the panel. Light entities don't update at all when operated from either the panel or elsewhere. Not sure if this helps at all...

Thanks for all your work on this. Love the Lovelace panel but not keen on Tasmota so this project is a real bonus for my setup.

HA-TB303, 2022-09-03:
On lights the "limit" of how many you can put in 1 gridcard seems to be 4. If I put 1-3 lights in the card the issues does not occur, it starts when adding the 4th light entity.

joBr99, 2022-09-04:
Okay, 
can flash current development version and test with this one?

`FlashNextion http://nspanel.pky.eu/lui.tft`

HA-TB303, 2022-09-05:
Uploaded the new firmware. The issue is still the same. You should be able to reproduce this with multiple light entities on the same gridcard. I tried this on a totally different HA instance in another network and I am having exactly the same issue.

joBr99, 2022-09-05:
Wasn't able to repoduce this with tasmota even with more than 4 entites.

Can you update the appdaemon backend to latest main, restart appdaemon and test again?

HA-TB303, 2022-09-05:
I have no newer appdeamon version available. 

[image]

Dit you test with esphome? I am aware it works fine with tasmota (a little slower though).

joBr99, 2022-09-05:
not appdaemon, the lovelace app in the hacs store

HA-TB303, 2022-09-05:
Also latest:

[image]

[image]

joBr99, 2022-09-05:
yeah, you need to update to the development version/main

[image]

HA-TB303, 2022-09-05:
Downloaded main, restarted appdeamon and I still have the same issue.

joBr99, 2022-09-05:
ok, my theory was that esphome is sending the messages to fast, causing a buffer overflow on nextion

with the dev version I implemented a dedub for the mqtt messages so the backend won't send the same messages twice, but apparrently this wasn't the issue, I'm out again no idea

joBr99, 2022-09-05:
@sairon is esphome logging unexpected serial output from the panel?

in case there is an error on the nextion screen it will return error codes, for example 0x24 for buffer overflows

```
22:37:45.211 MQT: tele/tasmota_NsPanelTerrasse/RESULT = {"nextion":"bytes('24')"}
```

sairon, 2022-09-06:
@joBr99 If it's not prefixed with `55BB` (hex), then yes, it should return false [here](https://github.com/sairon/esphome-nspanel-lovelace-ui/blob/dev/components/nspanel_lovelace/nspanel_lovelace.cpp#L40) and then be logged [here](https://github.com/sairon/esphome-nspanel-lovelace-ui/blob/dev/components/nspanel_lovelace/nspanel_lovelace.cpp#L27).

peetereczek, 2022-09-29:
I can reproduce the issue with release 3.4.0 of the backend component and NS Panel screens, but Ididn't spotted it on latest dev branch and with NS Panel screen from the link in comment (http://nspanel.pky.eu/lui.tft)

I was testing with 4 light on one card and alarm icon on screensaver

joBr99, 2022-09-29:
If I remember correctly our latest result was that it only occures with more than 4 lights.

peetereczek, 2022-09-29:
> If I remember correctly our latest result was that it only occures with more than 4 lights.

Aha, I see, so I have a bit different setup with lights

Anyway I had a glitch on screensaver with alarm state icon (top left corner) which gone, while using dev branch codes

sairon, 2022-11-24:
I was trying to look into this issue tonight but I still don't have an idea why this is happening. I've hooked up a logic analyzer to the TFT pins to see what's going on when this issue happens but I don't see any data corruption on the serial line. Here's a short session when trying to reproduce the issue - toggling the light "houby" from the HA web three times - states should go ON->OFF->ON, but after the last change the icon stayed blue:
```
[21:58:21][D][nspanel_lovelace:106]: Sending custom command: entityUpd~Obývák~1|1~light~light.lustr~~17299~Lustr~0~light~light.lustr_nocni~~17299~Lustr noční~0~light~light.lustr_rgb~~17299~Lustr RGB~0~switch~switch.hifi~~17299~HiFi~0~light~light.houby~~65222~Houby~1~light~light.rgb~~17299~RGB~0
[21:58:21][D][nspanel_lovelace:106]: Sending custom command: entityUpdateDetail~light.houby~~65222~1~disable~disable~disable~Color~Teplota barvy~Jas~disable
[21:58:22][D][nspanel_lovelace:106]: Sending custom command: entityUpd~Obývák~1|1~light~light.lustr~~17299~Lustr~0~light~light.lustr_nocni~~17299~Lustr noční~0~light~light.lustr_rgb~~17299~Lustr RGB~0~switch~switch.hifi~~17299~HiFi~0~light~light.houby~~17299~Houby~0~light~light.rgb~~17299~RGB~0
[21:58:22][D][nspanel_lovelace:106]: Sending custom command: entityUpdateDetail~light.houby~~17299~0~disable~disable~disable~Color~Teplota barvy~Jas~disable
[21:58:23][D][nspanel_lovelace:106]: Sending custom command: entityUpd~Obývák~1|1~light~light.lustr~~17299~Lustr~0~light~light.lustr_nocni~~17299~Lustr noční~0~light~light.lustr_rgb~~17299~Lustr RGB~0~switch~switch.hifi~~17299~HiFi~0~light~light.houby~~65222~Houby~1~light~light.rgb~~17299~RGB~0
[21:58:23][D][nspanel_lovelace:106]: Sending custom command: entityUpdateDetail~light.houby~~65222~1~disable~disable~disable~Color~Teplota barvy~Jas~disable
```
The same can be seen on the serial line - notice that both first two and last two rows (change to the ON state) are identical. There is no unexpected output received on the TX line (i.e. from the Nextion FW).
```
55 BB 00 01 65 6E 74 69 74 79 55 70 64 7E 4F 62 C3 BD 76 C3 A1 6B 7E 31 7C 31 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 6E 6F 63 6E 69 7E EE BD A4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 6E 6F C4 8D 6E C3 AD 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 72 67 62 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 52 47 42 7E 30 7E 73 77 69 74 63 68 7E 73 77 69 74 63 68 2E 68 69 66 69 7E EE 93 82 7E 31 37 32 39 39 7E 48 69 46 69 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E EE 9F 9E 7E 36 35 32 32 32 7E 48 6F 75 62 79 7E 31 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 72 67 62 7E EE 8F 97 7E 31 37 32 39 39 7E 52 47 42 7E 30 67 19
55 BB 5F 00 65 6E 74 69 74 79 55 70 64 61 74 65 44 65 74 61 69 6C 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E 7E 36 35 32 32 32 7E 31 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 43 6F 6C 6F 72 7E 54 65 70 6C 6F 74 61 20 62 61 72 76 79 7E 4A 61 73 7E 64 69 73 61 62 6C 65 9B F8
55 BB 00 01 65 6E 74 69 74 79 55 70 64 7E 4F 62 C3 BD 76 C3 A1 6B 7E 31 7C 31 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 6E 6F 63 6E 69 7E EE BD A4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 6E 6F C4 8D 6E C3 AD 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 72 67 62 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 52 47 42 7E 30 7E 73 77 69 74 63 68 7E 73 77 69 74 63 68 2E 68 69 66 69 7E EE 93 82 7E 31 37 32 39 39 7E 48 69 46 69 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E EE 9F 9E 7E 31 37 32 39 39 7E 48 6F 75 62 79 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 72 67 62 7E EE 8F 97 7E 31 37 32 39 39 7E 52 47 42 7E 30 AD F6
55 BB 5F 00 65 6E 74 69 74 79 55 70 64 61 74 65 44 65 74 61 69 6C 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E 7E 31 37 32 39 39 7E 30 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 43 6F 6C 6F 72 7E 54 65 70 6C 6F 74 61 20 62 61 72 76 79 7E 4A 61 73 7E 64 69 73 61 62 6C 65 1B D5
55 BB 00 01 65 6E 74 69 74 79 55 70 64 7E 4F 62 C3 BD 76 C3 A1 6B 7E 31 7C 31 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 6E 6F 63 6E 69 7E EE BD A4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 6E 6F C4 8D 6E C3 AD 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 6C 75 73 74 72 5F 72 67 62 7E EE 8C B4 7E 31 37 32 39 39 7E 4C 75 73 74 72 20 52 47 42 7E 30 7E 73 77 69 74 63 68 7E 73 77 69 74 63 68 2E 68 69 66 69 7E EE 93 82 7E 31 37 32 39 39 7E 48 69 46 69 7E 30 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E EE 9F 9E 7E 36 35 32 32 32 7E 48 6F 75 62 79 7E 31 7E 6C 69 67 68 74 7E 6C 69 67 68 74 2E 72 67 62 7E EE 8F 97 7E 31 37 32 39 39 7E 52 47 42 7E 30 67 19
55 BB 5F 00 65 6E 74 69 74 79 55 70 64 61 74 65 44 65 74 61 69 6C 7E 6C 69 67 68 74 2E 68 6F 75 62 79 7E 7E 36 35 32 32 32 7E 31 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 64 69 73 61 62 6C 65 7E 43 6F 6C 6F 72 7E 54 65 70 6C 6F 74 61 20 62 61 72 76 79 7E 4A 61 73 7E 64 69 73 61 62 6C 65 9B F8
```

I might revise this issue later, possibly compare Tasmota end ESPHome setups, but for the time being, I've added a workaround that might help getting rid of this annoyance. It exploits the fact that the panel pages are stateless, so transferring all the commands twice should not cause any issues (please correct me if I'm wrong, @joBr99). This workaround is on by default, but can be disabled by setting `use_missed_updates_workaround` option to `false`.

HA-TB303, 2022-11-25:
Ok, I will test this later this weekend. Thanks!

sairon, 2022-11-25:
Had a moment of enlightenment today and realized that the root cause might be in the timing. Turns out the ESPHome component is actually too fast when compared with Tasmota :nerd_face:

I confirmed this theory by taking another look at the waveforms on the logic analyzer, finding out that the two messages sent right after another when light status is changed (`entityUpd` and `entityUpdateDetail`) are looking almost like a single message.  This was also reason why switches were not affected, because there is no `entityUpdateDetail` issue for them. After adding a short delay to the MQTT message callback, the issue is gone, which I validated using a simple stress test mocking fast subsequent status updates. The previous workaround improved the situation, yet there were still some missed events. I also checked that with Tasmota there's always ~100 ms "natural" delay between messages, but "artificial" 75 ms wait in my component seems to works as well. It seems that complex `on_incoming_message` callbacks can also introduce such a delay, mitigating the issue, so it still kinda makes sense to add an option for disabling this delay.

@joBr99 It seems that sometimes the Nextion firmware can't cope with messages sent shortly after another (still had some issues with 50 ms delays, 75 ms and more seems to be working), ignoring the first `entityUpd`, or at least a part of it (could be the cause why this happens only with >3 lights). There always needs to be some delay between messages, although the protocol doesn't define such requirement.

joBr99, 2022-11-25:
Yes, this is something that I also thought of, maybe we are to fast for the berry driver and we are fillig up the serial buffer faster than the stm32 can handle.

HA-TB303, 2022-11-26:
Hi, I have tried this. Made sure the panel has the latest display firmware (http://nspanel.pky.eu/lui-release.tft) the lastet apdeamon version and the lastet version of nspanel lovelace UI. Then I cleaned the build files in esphome and reflashed (OTA) the esphome firmware on the panalen using this config;

```
substitutions:
  devicename: nspanelbadk
  friendly_devicename: NSPanel Badkamer
  panel_recv_topic: "tele/nspanelbadk/RESULT"
  panel_send_topic: "cmnd/nspanelbadk/CustomSend"

esphome:
  name: $devicename

esp32:
  board: esp32dev

wifi:
  networks:
    - ssid: !secret esphome_wifi_ssid
      password: !secret esphome_wifi_password

# Logger. Disable the temperature sensor etc. to focus on the HMI development
logger:
  level: DEBUG
  logs:
    sensor: WARN
    resistance: WARN
    text_sensor: WARN
    ntc: WARN

ota:
  safe_mode: true

# API. Add api_pwd to your secrets.yaml.
api:
  services:
    - service: upload_tft
      variables:
        url: string
      then:
        - lambda: |-
            id(nspanel).upload_tft(url);
    # Service to play a rtttl tone
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda "return song_str;"
    # Service to send a command directly to the display. Useful for testing
    - service: send_command
      variables:
        cmd: string
      then:
        - lambda: "id(nspanel).send_custom_command(cmd.c_str());"
    # Service to pusblish to mqtt used for refreshin the panel
    - service: publish_to_recv_topic
      variables:
        cmd: string
      then:
        - mqtt.publish:
            topic: $panel_recv_topic
            payload: !lambda "return cmd;"
    # Service to send a command wake the screen
    - service: wake
      then:
        - lambda: 'id(nspanel).send_custom_command("wake");'
    # Service to send a command to show screensaver (some of the values are hardcoded)
    - service: show_screensaver
      then:
        - mqtt.publish:
[... 163 lines cut ...]
binary_sensor:
  # Left button below the display
  - platform: gpio
    name: $friendly_devicename Left Button
    pin:
      number: GPIO14
      inverted: true
    id: left_button
    on_press:
      then:
        - switch.turn_on: relay_1

  # Right button below the display
  - platform: gpio
    name: $friendly_devicename Right Button
    pin:
      number: GPIO27
      inverted: true
    id: right_button
    on_press:
      then:
        - switch.toggle: relay_2
```

It responds a little better now, but I am still missing a lot of status updates.

sairon, 2022-12-01:
OK, reopening this ticket so I can gather more feedback. If status misses happen in your setup, please send here the following:
1. Full ESPHome config of your NSPanel (obviously without secrets :wink:).
2. Capture of the MQTT traffic when the issue occurs (with timestamps), e.g. from a similar command:
`
mosquitto_sub -h YOUR_MQTT_BROKER_IP -t "cmnd/nspanel/CustomSend" -t "tele/nspanel/RESULT" | ts "%.s"
`
3. (optional, but could also help) Debug log of the NSPanel ESPHome device.

@HA-TB303 Could you please try to reproduce it again and attach the missing logs? Just with the device configuration alone I am not able to reproduce it in my environment.

sairon, 2023-02-02:
There's been no activity for the past two months and I haven't encountered the issue since implementing the workaround either. Closing it as resolved for now.

commit ff34a3ea (2022-11-24): feat: add use_missed_updates_workaround option

Fixes #8
closed 2023-02-02 by sairon