sairon/esphome-nspanel-lovelace-ui #9: lovelace-ui not showing on the nspanel with ESPHome flashed

Videothek, 2022-08-17:
Hello,

first of all thank you for your effort to make the lovelace-ui available for the ESPHome Firmware on the nspanel.

Now to my problem:

I flashed my nspanel with ESPHome today.
I was following [this](https://www.espthings.io/index.php/2022/08/07/nspanel-lovelace-ui-a-great-add-on-for-home-assistant/) Guide.
So i flashed the config suggested (yours) on the nspanel and made the necessary changes to it such as mqtt credentials or panel name.
My nspanel shows as online in the ESPHome integration and can comunicate with MQTT (as far as i can tell from the Log output from ESPHome).
Appdaemon can communicate with MQTT too (also read from Logs).
But somehow the nspanel doesn´t show the lovelace-ui i configured (also used the example from the Guide), it just shows the standard screen.

Because i am a noobie to flashing my own devices with costum firmware, i dont know how to troubleshoot these things properly.
So if you need any logs, just tell me and i will provide them here.

Thank you in advance :-).

woody166, 2022-08-17:
Hi, did you upload the tft file from Services in HA using the url from http://nspanel.pky.eu/lui-release.tft or from the console of your nspanel using FlashNextion http://nspanel.pky.eu/lui-release.tft

Videothek, 2022-08-17:
> Hi, did you upload the tft file from Services in HA using the url from http://nspanel.pky.eu/lui-release.tft or from the console of your nspanel using FlashNextion http://nspanel.pky.eu/lui-release.tft

No, i havent.
But funny story, i am uploading it right now.
I am sorry to bother you with this issue, but maybe its helpful for someone else that also cant read the ReadMe like me🤦.
But thank you very much for your fast reply 😄.

But now its saying: "Waiting for content..." and "This is taking longer than usual, please check your background configuration".
Any guess, what i should change?

Videothek, 2022-08-17:
Ok i found my error, i was missing the app from the HACS store.

Now everything is working as it should.

Thank you very much 😃.

closed 2022-08-17 by Videothek

JasperHorn, 2022-12-31:
Just wanted to mention my appreciation for this issue (even though it's more of a support request). I was hanging my head against the wall as I was simply unable to understand the need for this step from the esphome guides. Then I found this issue and finally understood. Thanks!