sairon/esphome-nspanel-lovelace-ui #11: Feature request: Decouple buttons from controlling power outputs

jrot, 2022-09-25:
I use my NsPanel mainly for controlling smart lights and media players,
I would to decouple the buttons from the relays since I don't use them (yet).

This is possible in Tasmota:
https://docs.nspanel.pky.eu/phys-btn/

Would it be possible to integrate in the ESPHome version as well?

randybb, 2022-09-25:
What? The benefit of this integration is that your ESP32 is just an ESP32 with ESPHome, so you can do whatever you want to do with it. This components only adds a bridge between MQTT data send by AppDaemon and NSPanel, nothing else.
For example in my office I am controlling two lights from these buttons. Left is an intelligent one controlled by HA (yeelight), Right is controlling dumb recessed lights through integrated relay. The mqtt publish is used to open the right page on NSpanel where I can see which light I have toggled and in case of an intelligent lamp I can change its parameters...
```yaml
binary_sensor:
  - platform: gpio
    name: ${name} Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - homeassistant.service:
          service: light.toggle
          data:
            entity_id: light.office_r_desk_strip
      - mqtt.publish_json:
          topic: $panel_recv_topic
          payload: |-
            root["CustomRecv"] = "event,buttonPress2,navigate.cardGrid_main,button";
  - platform: gpio
    name: ${name} Right Button
    pin:
      number: 27
      inverted: true
    on_click:
      - switch.toggle: relay_1
      - mqtt.publish_json:
          topic: $panel_recv_topic
          payload: |-
            root["CustomRecv"] = "event,buttonPress2,navigate.cardGrid_main,button";
```

jrot, 2022-09-25:
@randybb thank you for your comment. I need more coffee.
I see now that this is already possible :facepalm: 

I will close this ticket, thanks for the fast response :+1: 

And thanks to @sairon for making this UI work with ESPHome :+1:

closed 2022-09-25 by jrot