sairon/esphome-nspanel-lovelace-ui #14: [Feature Request] or [Question] Current card tracking for custom button actions on each card

marek-vrana, 2022-10-28:
Hi guys, this project is spectacularly awesome! I was a bit afraid of going with esphome way instead of tasmota in the beginning. Can say it was worth it! Configuration is so easy and options are almost infinite. Thank you for your time you're spending on this project!

Since I'm still looking around, I was hoping if there is a possibility to use different button actions on each of cards. 
My idea is: 
* when I'm on cover details card, I could use physical buttons to tilt the covers (only by software)
* light details card will allow you dim the light by long pressing the buttons, short press left will set lowest brightness, short press right will set max brightness (also only y software)
* when on home card (default one), I could use buttons to switch the relays (hardware buttons)

Maybe adding a current card to nspanel state would allow to implement it?
Are there any limitations in ESPHome or MQTT that I'm not aware of or is it doable?

sairon, 2022-11-25:
You should be able to do this by adding a custom parser to the `on_message` of the MQTT client. This way you can extract shown page type (from `pageType` message sent by the AppDaemon) and page name (from `entityUpd` message sent afterwards). See the [README](https://github.com/sairon/esphome-nspanel-lovelace-ui/blob/f945e272d7dd70849e245dc9f23955f80c4c4a5f/README.md#usage) for an older revision which used `on_message` for passing of the MQTT messages to the display. The extracted value can be saved to a [global variable](https://esphome.io/guides/automations.html#global-variables) which can be used in your button automations.

I'm leaving this issue open with the possibility to add this parsing to the `nspanel_lovelace` component itself as you suggest, with no ETA promised :)

aussiebum71, 2023-02-16:
I think this would be a fantastic addition. Some users have commented that using the brightness slider on light entities that support brightness, is too small of an area to slide with your finger. Being able to adjust the brightness with the buttons
or using them to open or close the cover, if a cover entity is showing is a great idea, 

The buttons have multiple functions depending on what entity is displayed.