sairon/esphome-nspanel-lovelace-ui #18: Entities below "cardMedia" not showing up

Shogun1978, 2022-12-20:
Got the NSPanel up and running, great work, thank you so far!

Currently I am struggling with the entities below the cardMedia. The media player is showing and everything is working, but no items below:

```
      - type: cardMedia
        title: Büro OG
        entity: media_player.bar
        entities:
          - entity: light.buro_og_deckenlicht
          - entity: light.buro_og_schreibtisch
          - entity: light.buro_og_standlich
```

I followed this description: [https://docs.nspanel.pky.eu/card-media/](url). In the end, I want to call a script to switch trough different playlists/radio stations. Is there something I'm doing wrong?

sairon, 2023-01-27:
Just tried a similar configuration on my instance and it works as expected. Given how the NSPanel with custom firmware works, it's unlikely such problem could be caused by the ESPHome component, but rather it's something in your setup. Make sure all components (especially the AppDaemon backend and TFT firwmare) are up to date. Eventually you may also try using the official Tasmota build to rule out the ESPHome component is causing it. For now I'm closing the issue, let me know if you find something that suggests it should be reopened.

closed 2023-01-27 by sairon

joBr99, 2023-01-28:
defiantly nothing related to esp home, as esp home is only relaying the messages :)