sairon/esphome-nspanel-lovelace-ui #19: upload_tft How?

DrJeff, 2022-12-27:
I used the service inside of HA but in GUI there is no where  to add the URL in YAML I can add the URL with quotes or without no joy, if I add a Key of "url" and the URL still no joy what am I missing?
Thanks

DrJeff, 2022-12-27:
Not sure how or why the service call did not work. For a work around I added  this to the actual button and updated the panel that way. 
```
switch:
  # Physical relay 1
  - platform: gpio
    name: $friendly_devicename Relay 1
    id: relay_1
    pin:
      number: 22
    on_turn_on:
      - lambda: |-
                id(nspanel).upload_tft([url](http://nspanel.pky.eu/lui-us-p-release.tft));
```

randybb, 2022-12-27:
One of the option is to use the service _upload_tft_ mentioned in https://github.com/sairon/esphome-nspanel-lovelace-ui and run it with url from dev tools
[image]

DrJeff, 2022-12-28:
Ok I must have had the yaml spacing wrong. I was not getting the variable for `url` but now I do thanks.

closed 2022-12-28 by DrJeff