sairon/esphome-nspanel-lovelace-ui #20: Can not upload the TFT file

lyazide, 2023-01-03:
I am using the app on a EU nspanel with the ESPhome soltware.
When I try to upload the software by calling the service (url = http://nspanel.pky.eu/lui-release.tft):

```
api:
  services:
    - service: upload_tft
      variables:
        url: string
      then:
        - lambda: |-
            id(nspanel).upload_tft(url);
```

I get the following message:
```
[14:04:53][D][nspanel_lovelace_upload:150]: Connected
[14:04:53][D][nspanel_lovelace_upload:156]: Requesting URL: http://nspanel.pky.eu/lui-release.tft
[14:04:53][W][nspanel_lovelace_upload:166]: HTTP Request failed; URL: http://nspanel.pky.eu/lui-release.tft; Error: connection refused, retrying (1/5)
[14:04:54][W][nspanel_lovelace_upload:166]: HTTP Request failed; URL: http://nspanel.pky.eu/lui-release.tft; Error: connection refused, retrying (2/5)
[14:04:54][W][nspanel_lovelace_upload:166]: HTTP Request failed; URL: http://nspanel.pky.eu/lui-release.tft; Error: connection refused, retrying (3/5)
[14:04:54][W][nspanel_lovelace_upload:166]: HTTP Request failed; URL: http://nspanel.pky.eu/lui-release.tft; Error: connection refused, retrying (4/5)
[14:04:55][W][nspanel_lovelace_upload:166]: HTTP Request failed; URL: http://nspanel.pky.eu/lui-release.tft; Error: connection refused, retrying (5/5)
[14:04:55][D][nspanel_lovelace_upload:274]: Restarting Nextion
```

I can download the file so the link is not dead.

randybb, 2023-01-03:
Do you have configured dns, gateway (if you are not using DHCP) in esphome and can it reach the Internet via your firewall/router?

lyazide, 2023-01-03:
Hi,
Thanks for your reply.
Here is what I set setup:

```
manual_ip:
    static_ip: 192.168.1.xx
    gateway: 192.168.1.1
    subnet: 255.255.255.0
```

randybb, 2023-01-03:
so you are using static config, you need to add dns https://esphome.io/components/wifi.html
```yaml
  manual_ip:
    dns1: 192.168.1.1
```

lyazide, 2023-01-03:
Great, I am so stupid ;) Thanks that solve the issue. I missed that one.

lyazide, 2023-01-03:
For manual setup we have to set dns1:

manual_ip:
    static_ip: 192.168.1.xx
    gateway: 192.168.1.1
    subnet: 255.255.255.0
dns1: 192.168.1.1

closed 2023-01-03 by lyazide