sairon/esphome-nspanel-lovelace-ui #21: upload_tft fails at 86% when bluetooth_proxy is enabled

rschueler, 2023-01-08:
Thank you for building this great component. I prefer using it over Tasmota because I am a big fan of ESPhome and I can use my NSpanels to extend the Bluetooth range of my home assistant instance throughout the house. 

However, enabling the bluetooth_proxy and ble_tracker components on the device has a downside. Whenever I try to upgrade the panel TFT firmware, the progress bar becomes stuck at 86%. This issue is reproducible with the latest builds of ESPhome and also with TFT firmware version 3.8.3. 

As a workaround, I have commented out both Bluetooth components in my ESPhome configuration and am able to successfully update the TFT firmware without issue. However, I am concerned that this workaround may reduce the lifespan of the ESP32 flash memory if I have to overwrite it twice for each TFT update. Is there a way that I can help you find the root cause of this problem?

joshcliffejones, 2023-01-10:
> Thank you for building this great component. I prefer using it over Tasmota because I am a big fan of ESPhome and I can use my NSpanels to extend the Bluetooth range of my home assistant instance [...]

Im so glad i stumbled across this as i have also had problems with the TFT Upload feature using ESPHome while having the Bluetooth proxy enabled.

I attempted an upgrade to the latest TFT build and it just stuck at 0% for me. Having read this, i have now done the same and commented out the bluetooth proxy component and now the update appears to be working as it was before!

A shame, yes. But thankfully not critical for me as i will just fully remove the proxy, but still would be cool if it was fixed. :)

sairon, 2023-01-27:
I am not using the `bluetooth_proxy` component so I haven't encountered the issue before, but I confirm I can reproduce it perfectly when I enable it. The problem is fundamentally caused by the same issues mentioned in the [red box in the official docs](https://esphome.io/components/bluetooth_proxy.html). With Bluetooth proxy enabled, the upload fails after few percent with stack trace that usually suggests there's been problem allocating memory.

Fortunately I found out there's an easy fix - adjust the `upload_tft` lambda so it disables the BLE scan before the upload is started. With that change, I've been always able to finish the upgrade without errors. To do that, simply add an ID to the `esp32_ble_tracker` component (it's added implicitly with `bluetooth_proxy`, so if you don't have it in your configuration, just add it), e.g.:

```
esp32_ble_tracker:
  id: ble_tracker
```

and then modify the `upload_tft` action to call the `stop_scan` action on it before the upload is started:

```
    - service: upload_tft
      variables:
        url: string
      then:
        - lambda: |-
            id(ble_tracker).stop_scan();
            id(nspanel).upload_tft(url);
```

Also if you initially flashed your NSPanel with ESPHome version older than 2022.12.0, make sure to do at least one upload using the serial connection, so the new partition schema can be applied. In docs it's also recommended not to use the `web_server` component but I've been able to do a successful TFT firmware upgrade even with that but YMMV.

@rschueler @joshcliffejones Can you let me know if it works for you too? I will make sure this quirk is mentioned in the documentation then.

rschueler, 2023-01-27:
Thank you for your assistance. The solution works well for me with this small addition. Is it possible for me to call the function "id(ble_tracker).start_scan()" immediately after "upload_tft," or do I need to manually enable the scan once the display has been flashed due to upload_tft being asynchronous?

sairon, 2023-01-27:
Thanks for your feedback! I will add a note to the documentation.

> or do I need to manually enable the scan once the display has been flashed due to upload_tft being asynchronous?

There's no need to do so, because once `upload_tft` action finishes, it does a soft restart of the display and ESPHome as well, so the scanning starts again.

commit 4bffa2b7 (2023-02-02): chore: add info about issues caused by Bluetooth components

Fixes #21
closed 2023-02-02 by sairon