sairon/esphome-nspanel-lovelace-ui #22: New full screen notifications

mruizh, 2023-02-02:
Hi ,
 I have updated to NsPanel Lovelace v.3.9.4 and tested some new features with last tft firmware updated in Nspanel. I can confirm that new functions like screensaver runs directly in apps.yaml and with little changes in nspanel.yaml, Full Screen Notification works with icon, colors and font sizes. 

This is my nspanel.yaml.

    # Service to send a command To show the screen of a specific like
    # interaction of the buttons are on implemented yet
    # New Variables icon . Not all implemented. Icon must be "U" value (copy and paste).
    - service: notify_fullscreen
      variables:
        title: string
        description: string
        button1: string
        button2: string
        time_secs: string
        icon: string
      then:
        # show notification screen
        - lambda: 'id(nspanel).send_custom_command("pageType~popupNotify");'
        # set values on notification screen
        # color defined as number created in binary and converted to decimal (rrrrggggbbbbaaaa)
        #- lambda: 'id(nspanel).send_custom_command("entityUpdateDetail~id~"+ title + "~65535~" + button1 +"~3840~" + button2 + "~61440~" + description + "~65535~" + time_secs);'
        # New funcionalities v 3.9.4
        #entityUpdateDetail~internalName~heading~headingColor~button1text~button1color~button2text~tB2Color~notificationText~textColor~sleepTimeout~fontIdText~icon~iconColor

        - lambda: 'id(nspanel).send_custom_command("entityUpdateDetail~id~"+ title + "~65535~" + button1 +"~3840~" + button2 + "~61440~" + description + "~65535~" + time_secs + "~3~" + icon + "~65535~");'
        - rtttl.play: "scale_up:d=32,o=5,b=100:c,c#,d#,e,f#,g#,a#,b"
 
Enjoy it!

sairon, 2023-02-02:
Thanks for the heads-up! But since this is a tip rather than an issue, I'm closing it, next time please use the [discussions](https://github.com/sairon/esphome-nspanel-lovelace-ui/discussions) page.

closed 2023-02-02 by sairon

mruizh, 2023-02-03:
Yes! my fault.