sairon/esphome-nspanel-lovelace-ui #23: Use component with ESP-IDF framework

rschueler, 2023-02-03:
Hi, I am trying to utilize the bluetooth_proxy on my nsPanel, but the latest speed improvements are based on the ESP-IDF framework. When I configure my YAML file to use this framework, I encounter the following error during compilation: "This feature is only available with frameworks ['arduino']." Is there a solution to this problem?

Furthermore, I have run into some difficulties with the latest release of ESPhome and HA 2023.2. I have found that when using the 2022.12.8 firmware, the file size becomes excessively large when Bluetooth is enabled. The situation is becoming increasingly problematic as the API password is set to be deprecated and the recommended API encryption requires additional space.

sairon, 2023-02-06:
I already tried to port it, there are some parts of the code that would need to be rewritten, because they indeed use Arduino framework features, however the main blocker was that the ESP-IDF framework can not handle MQTT messages correctly. Today I created an issue reporting the misbehavior: https://github.com/esphome/issues/issues/4133

Once the issue is resolved there is still some work to be done on my end, however I can not continue until the problem is resolved. There are some workarounds possible (e.g. introducing an artificial delay between the messages in the backend) but that is not the ideal way to proceed.

Scags104, 2023-03-29:
Following - same issues. 

Thanks

PhiKapJames, 2023-07-28:
Looks like it was fixed last month.  What work needs to be done that is left?

sairon, 2023-09-19:
@PhiKapJames The blocker is the display FW upload code, which uses HTTPClient library that is Arduino only. I did not have much time recently to invest into the rewrite effort - but maybe it could change in the following months. Also currently I can only presume it will address the issues with space constraints, because there is a chance that linking the HTTP APIs in ESP IDF (which are AFAIK not used in any other ESPHome component) will also cause the bloat.

joBr99, 2023-11-20:
@sairon Looks like the nextion upload code got ported to esp-idf

https://github.com/esphome/esphome/pull/5667
https://github.com/esphome/esphome/tree/dev/esphome/components/nextion

sairon, 2023-11-27:
@joBr99 Nice, thanks for the heads-up! That should be fairly easy to port now, I will try to find a time slot for that soon.

sairon, 2023-12-01:
I have kicked off the ESP-IDF support in #33, I'll do some more testing and prepare a new release soon. Any other testers are welcome at this point :pray:

PR #33 (2024-08-02): feat: add ESP-IDF support
Add support for `esp-idf` framework, partially based on ESPHome PR esphome/esphome#5667 by @edwardtfn (thank you, Edward!). The upstream code has been refactored to share more when Arduino or ESP-IDF is used and to use baud rate switching used in this "fork".

This is currently still WIP, I will do a bit more refactoring and testing, but initial tests are looking good and there are indeed some resource savings when compared to Arduino, as seen with my testing config:

Arduino:
```
RAM:   [=         ]  14.6% (used 47928 bytes from 327680 bytes)
Flash: [=======   ]  71. [... cut]
closed 2024-08-02 by sairon