sairon/esphome-nspanel-lovelace-ui #25: Issue with QR Code Card

GraCed1983, 2023-04-14:
Dear Sairon, first of all many thanks for the great work on NSPanel! I have the issue that QR Code Card is always showing the QR Code with None (default setting) and not changing the picture. No idea what I am doing wrong, tried it with entries directly and also by template, but the QR is always the same.

randybb, 2023-04-14:
This code is just a bridge between mqtt and nspanel running on esphome. If you have any issues with nspanel lovelace ui, then post it here https://github.com/joBr99/nspanel-lovelace-ui/issues

GraCed1983, 2023-04-15:
Dear Roman

Many thanks for your feedback! Will adress it right.

Best regards
Cedric

sairon, 2023-05-29:
Hi @GraCed1983, given how the NSPanel with custom firmware works, it's unlikely such problem could be caused by the ESPHome component, but rather it's something in your setup. Make sure all components (especially the AppDaemon backend and TFT firwmare) are up to date. Eventually you may also try using the official Tasmota build to rule out the ESPHome component is causing it. For now I'm closing the issue, let me know if you find something that suggests it should be reopened.

closed 2023-05-29 by sairon