sairon/esphome-nspanel-lovelace-ui #26: Can only upload tft file with tasmota.

weltmeyer, 2023-05-21:
When i try to upload a tft file while using the esphome variant, i always get only to returned [].

i need to flash with tasmota using "FlashNextionAdv6 http://localhttpserver/tft/nspanel-v4.1.1.tft" which is working fine.

is there a way to implement the same command for esphome?
It is programmed here: https://github.com/joBr99/nspanel-lovelace-ui/blob/main/tasmota/autoexec.be 

`def flash_nextion_adv(cmd, idx, payload, payload_json)
    def task()        
        if idx==0
            nextion.flash_proto_version = 1
            nextion.flash_proto_baud = 921600
        elif idx==1
            nextion.flash_proto_version = 0
            nextion.flash_proto_baud = 921600
        elif idx==2
            nextion.flash_proto_version = 1
            nextion.flash_proto_baud = 115200
        elif idx==3
            nextion.flash_proto_version = 0
            nextion.flash_proto_baud = 115200
        elif idx==4
            nextion.flash_proto_version = 1
            nextion.flash_proto_baud = 256000
        elif idx==5
            nextion.flash_proto_version = 0
            nextion.flash_proto_baud = 256000
        elif idx==6
            nextion.ser.deinit()
            nextion.ser = serial(17, 16, 9600, serial.SERIAL_8N1)
            nextion.flash_proto_version = 0
            nextion.flash_proto_baud = 921600
        else
            nextion.flash_proto_version = 0
            nextion.flash_proto_baud = 115200
        end
        
        nextion.flash_nextion(payload)
    end`

idx=6 is the only one working for me :(
It seems like they are resetting the serial port and start with 9600 baud, then uploading with 921600.
also they are using a slightly different command (proto_version=0) when initiating the upload
using the espcomponent and setting UART to 9600 doesnt work for me.

jdenda, 2023-05-22:
Hey, 

I have a similar problem, it was possible to upload the first TFT firmware via esphome_upload_tft as shown in the example. But now when I try to upgrade the firmware via the same function I get the following error: 

> [20:27:34][D][nspanel_lovelace_upload:150]: Connected
[20:27:34][D][nspanel_lovelace_upload:156]: Requesting URL: http://192.168.1.166:8123/local/nspanel/lui-release.tft
[20:27:34][D][nspanel_lovelace_upload:188]: Updating Nextion
[20:27:34][D][nspanel_lovelace:119]: Sending: whmi-wris 8261216,921600,1
[20:27:34][C][uart.arduino_esp32:077]: Setting up UART...
[20:27:34][D][nspanel_lovelace_upload:213]: Waiting for upgrade response
[20:27:36][D][nspanel_lovelace_upload:218]: Upgrade response is []
[20:27:36][D][nspanel_lovelace_upload:223]: preparation for tft update failed 0 ""
[20:27:36][D][nspanel_lovelace_upload:276]: Restarting Nextion
[20:27:36][D][nspanel_lovelace:119]: Sending: rest
[20:27:38][D][nspanel_lovelace_upload:279]: Restarting esphome

Sometimes I get this variant: 

> [20:37:34][D][nspanel_lovelace_upload:150]: Connected
[20:37:34][D][nspanel_lovelace_upload:156]: Requesting URL: http://nspanel.pky.eu/lui-release.tft
[20:37:34][D][nspanel_lovelace_upload:188]: Updating Nextion
[20:37:34][D][nspanel_lovelace:119]: Sending: whmi-wris 8261216,921600,1
[20:37:34][C][uart.arduino_esp32:077]: Setting up UART...
[20:37:34][D][nspanel_lovelace_upload:213]: Waiting for upgrade response
[20:37:34][D][nspanel_lovelace_upload:218]: Upgrade response is [1A]
[20:37:34][D][nspanel_lovelace_upload:223]: preparation for tft update failed 26 ""
[20:37:34][D][nspanel_lovelace_upload:276]: Restarting Nextion
[20:37:34][D][nspanel_lovelace:119]: Sending: rest
[20:37:36][D][nspanel_lovelace_upload:279]: Restarting esphome

brotherseb, 2023-05-29:
Same problem for me :(

sairon, 2023-05-29:
Hi everyone. Can you post some more details about your setup - especially the complete ESPHome device config? First I thought there might be something in the latest ESPHome or NSPanel firmware, but tried updating the TFT display few times back and forth between 4.0.5 and 4.1.1 with ESPHome 2023.5.4 and I didn't manage to reproduce the behavior you're reporting. Aren't you using any Bluetooth components? They are [known to cause trouble](https://github.com/sairon/esphome-nspanel-lovelace-ui#coexistence-with-bluetooth-components). Also you can try setting the `update_baud_rate` to `115200` to check if the upload works at least then. Furthermore, the upgrade sometimes only works if the device correctly switches to/from the reparse mode - i.e. if the first update fails, you should power-cycle the display before another try.

sairon, 2023-05-29:
I see, thanks for the clarification. This happens when the re-flashing of the original stock firmware (or other firmware using different baud rate) fails, as explained here: https://github.com/joBr99/nspanel-lovelace-ui/issues/862#issuecomment-1510483928

Otherwise the display should initialize at 115200 even when the upload failed before. Simulating FlashNextionAdv6 could be done by setting the the `baud_rate` in the `uart` component itself to `9600` (i.e. [here](https://github.com/sairon/esphome-nspanel-lovelace-ui/blob/dev/examples/basic-config.yml#LL117C5-L117C14)).

Maybe there could be more scenarios that could lead to the display comm stuck at 9600 baud, but I will need more context what was happening before the device was brought to this state. Can everyone having the same issue report what's written on the display when the upload fails?

bennodepenno, 2023-06-03:
Hi,

I'm experiencing the same problem as described above. I'm using a white EU version of the nspanel. Tried changing the uart baudrate to 9600 and 921600, all result in the same effect as described by jenda. I'm not sure how update_baud_rate can be defined though. If it's possible to specify an update_baud_rate, please provide an example and I can try to see if it works.

mitekpl, 2023-07-01:
I have the same issue with upload tft when compiling esphome project with:
```
logger:
  baud_rate: 0
  level: INFO
```
I do not see any issues with update (even with update_baud_rate: 921600) when I switch back to logger level set to DEBUG.
Tried to modify nspanel_lovelace component source to switch all logs from ESP_LOGD to ESP_LOGI however it does not help.
I think the issue is deeper in ESPHOME optimization compiled without logs.

weltmeyer, 2023-08-25:
@sairon 
as i am now right before doing the latest update.. what exactly should i do to give you required informations before i start the update of the ha-component?
i am holding it back now so you can give me instructions for all the debug infos you may need.

sebPomme, 2023-12-01:
Hello, 
Looks like I have the same problem, I tried different `nspanel_lovelace.update_baud_rate:` (9600, 115200, 921600) nothing seem to work. 
For flashing the nextion I rolled back to my old esphome firmware (witch use `uart.baud_rate: 115200`) maybe we can find a solution somewhere.
BTW my screen:
```
[11:58:03][C][nextion:123]: Nextion:
[11:58:03][C][nextion:124]:   Device Model:     NX4832F035_011C
[11:58:03][C][nextion:125]:   Firmware Version: 48
[11:58:03][C][nextion:126]:   Serial Number:    XXXXX
[11:58:03][C][nextion:127]:   Flash Size:       16777216
[11:58:03][C][nextion:128]:   Wake On Touch:    True
```