sairon/esphome-nspanel-lovelace-ui #30: Component mqtt took a long time for an operation

Ulrar, 2023-08-18:
Hi,

Recently I've found my screens to be sluggish, sometimes taking ten or twenty seconds before coming out of screen saver.

Looking into it a bit, I noticed both are logging the following for every mqtt message :

```
[11:35:26][W][component:204]: Component mqtt took a long time for an operation (0.10 s).
[11:35:26][W][component:205]: Components should block for at most 20-30ms.
```

May not be related to this project at all, I'm just wondering if someone else has experienced this behavior with their NSPanel and might be seeing similar messages ? My other esp devices, as far as I can tell, aren't affected, just the nspanels.

Thanks

randybb, 2023-08-18:
Loglevel for these messages has been changed so it is visible when logging is set to warning, which is right.

We have a issue opened for this https://github.com/esphome/issues/issues/4717 where we are gathering list of components that need to be fixed (move from blocking delay to non blocking interrupt based delay, do some optimizations,...). But it this case it might be related to slow AppDaemon...

In any case, this is not related to this project.

sairon, 2023-09-19:
@Ulrar You can indeed ignore these messages, they are harmless and caused by how the data is passed from MQTT to the UART of the Nextion panel. The issues you are seeing are caused by something else - in that case it's good to correlate MQTT logs coming from the AppDaemon with the data seen in the ESPHome logs. If there indeed is something that is not propagated to the display, then either file a bug here or on the display FW repo.

closed 2023-09-19 by sairon