sairon/esphome-nspanel-lovelace-ui #35: UI is slow

twasilczyk, 2024-01-24:
I tried using ESPHome NSPanel Lovelace UI and what threw me off was how slow it felt.

To put real numbers to that feeling, I compared this firmware to Blackymas' NSPanel_HA_Blueprint by recording test videos with a 240fps camera. Differences are not huge, but I think they cross the boundary between "barely noticeable" and "laggy".

Time to start screen re-draw after button press
-
NSPanel_HA_Blueprint: 40ms (with "Show pages while loading" option enabled)
esphome-nspanel-lovelace-ui: 330ms

This makes blueprint firmware feel super responsive and esphome-lovelace firmware super slow. In addition to latency problem, target areas for lovelace UI are small, so that I tend to miss them and have to re-tap. This doesn't happen too often on Blueprint UI.

Re-draw time
-
NSPanel_HA_Blueprint: 270ms (that's 310ms from button press)
esphome-nspanel-lovelace-ui: 260ms (that's 590ms from button press)

However, esphome-nspanel-lovelace-ui re-draw time is sometimes as long as 1300ms

Entity action (light on/off) round-trip time
-
NSPanel_HA_Blueprint: 500ms
esphome-nspanel-lovelace-ui: 700-900ms

I guess that's due to MQTT vs HA API latency and could be fixed with #34 (I wasn't able to confirm it though).

grericht, 2024-02-22:
is there a difference between lovelace ui on tasmota and esphome?

sairon, 2026-10-27:
By switching from AppDaemon (which is now deprecated since [v4.8.0](https://github.com/joBr99/nspanel-lovelace-ui/releases/tag/v4.8.0)) to HA app, the performance much improved. Anyway, the bottleneck never was this ESPHome component itself, rather the rest of the "stack", so closing this as not planned.

closed 2026-10-27 by sairon